j4velin/Pedometer #62: The record of steps is ridiculous!

youlansedeninan, 2015-01-09:
### My cell phone is nexus 5, so I installed Pedometer. Because it can use the hardware step detection sensor of the Nexus 5. The number of step in the Pedometer is 17,803. 
### In the same time, I wear a wearable band. The number of steps in the band is 2377.
### I set 0.95 meter per step in the Pedometer, So I walked 16.91285 km during several hours.  Is it ridiculous?
[image]
[image]

j4velin, 2015-01-09:
I dont think 17.800 steps are ridiculous

j4velin, 2015-01-12:
Did you do anything else with your phone? Going by car / train or something? Note that I have no influence on the sensors sensitivity

youlansedeninan, 2015-01-15:
I don't know why.But it seems become normal.
[image]
Last Thu. and Fri. the steps is unbelievable. But this week , it works good.

Jabe, 2015-01-19:
Hi there,

similar issue after upgrading to Lollipop. Step counting seems kind of random now. Each step is counted 1-100 times. Tried reboot, wipe data, reinstall, not resolved.

Been using this app for a while now (on S4 running CM11) and tracked about 1.5 Megasteps ;) , but decided to switch to a (unofficial) Lollipop-based Google Play Edition Rom. So obviously this could be Rom or Kernel related, but maybe an issue with Lollipop?

Cheers

j4velin, 2015-01-19:
Works fine on my Nexus 5 with AOSP Lollipop

Jabe, 2015-01-20:
I've been playing around with this. I checked out GPS Status Pro, which reads the HW-pedometer value as well and shows the same bad behaviour. Then I installed yet another HW-pedometer app and now all three apps magically count correctly. When I uninstall the new app, both this app and GPS Status immediately start counting wrong again. So, uninstall all pedometers and keep only GPS Status, counts badly, install the other pedometer, fixes it. So... yeah, shared resource acting weird, what a pleasure.

evsh, 2015-04-10:
@Jabe, I have exactly the same problem with S4 and CM 12.1. What this another HW-pedometer app is, please?

sniperle, 2015-04-28:
Having the same problem on LG G2 with CM 12, millions of steps are shown to me ;-)

[image]

matthiasbeyer, 2016-04-01:
Oneplusone here, had 300,000 steps yesterday. :curly_loop: ... after bringing my phone to low-power mode...

closed 2017-04-23 by j4velin

matthiasbeyer, 2017-04-23:
Can report: 17,8k steps is not that much. When going for a walk, you can get to 17k! My record one day is 21k. Normal days, though, I'm around 9-11k!

300k tough is a bug. If the app crashes, it reports wrong numbers!