j4velin/Pedometer #76: App seems to stop to stop counting when phone is in deep sleep or so

benediktg, 2015-10-26:
I have a Samsung Galaxy S5 with Cyanogenmod 12.1 nightly (from 24th October 2015). When I watch the app while the screen is on the steps are counted pretty precisely. But I have found out that the counting does not work anymore when there is no other app running in the background. It worked as I recorded a GPS track with Osmand or while counting steps with the other app called "Pedometer" from F-Droid (I'm using F-Droid as the repo for effectively every app on my phone) which also keeps the phone slightly busy.

Could it be that the step detector and counter does not function in deep sleep mode on some devices? Or could it have something to to with the circumstance that I have no GApps installed and removed some other proprietary elements with "freecyngn" (https://github.com/mar-v-in/freecyngn)? Is there any possibility to get more information by the use of the logs?

j4velin, 2015-10-26:
That's quite possible. The step sensor should, per definition in the Android docs, return the number of steps since the last reboot and not the number of steps while the device wasn't in deep sleep - but it is not unusual that Samsung doesn't give a f*** about the Android docu. Especially if the device does not have a dedicated sensor processor, the sensor might just shut down if the CPU goes to sleep. It seems to work fine on my S6, but that might just be the different hardware the S5 and S6 have. Google Play services or something should not be required for the step sensor to work. You can build the app in debug config to create some logging, but I don't think that will be very useful.

benediktg, 2015-10-26:
I see.

If there was an option in the app to enable something like a wakelock and ensure that the step detector and counter work, would the impact on the battery usage be similar to e. g. the other app called pedometer (https://f-droid.org/repository/browse/?fdid=name.bagi.levente.pedometer) which seems to use a wakelock? And would it in spite of this be reasonable to implement such an option as a workaround for devices like the S5?

cshlxm, 2015-11-02:
I had also found that the Pedometer had stopped counting while my device screen-off .My device is  Samsung Galaxy S4.
It seems that you can solve this problem like this (https://github.com/googlesamples/android-BatchStepSensor      " Likewise, batched sensors can be used in the background (when the CPU is suspended), which requires manually flushing the sensor event queue before it overflows,").

j4velin, 2015-11-11:
I don't think that batching would fix it as it just seems Samsung shuts down the steps sensor on your devices when the display is off (in contrast to what the Android docu says). I very much doubt that batching the sensor data would keep it alive.
Implementing a wakelock might be a fix but then the app would be just as power consuming as any other pedometer app. The intention of the app was to have (nearly) no impact on battery life or other system ressources

benediktg, 2016-01-09:
Now I have installed Cyanogenmod 13 and somehow the step sensor seems to work as of recently. There are steps counted even when the screen is turned off since a while.

closed 2017-04-23 by j4velin