j4velin/Pedometer #79: Export format: What kind of timestamp is the data?

matthiasbeyer, 2016-04-10:
Hey,

I want to use the data I can export with the export functionality to create some graphics for my walking behaviour. Though I don't know what kind of timestamp you are using. It is not Unix timestamp, as this

```bash
$ date --date "@1454281200000"
So 3. Mai 10:00:00 CEST 48054
```

shows (`1454281200000` is a timestamp from the export).

What kind of timestamp are you using?

j4velin, 2016-04-10:
Milliseconds since 1970

matthiasbeyer, 2016-04-10:
Why would you pick such a weird export format? Why not simple `<YYYY>-<MM>-<DD>`? Is there any technical reason for this?

j4velin, 2016-04-10:
I dont think it weirder than any other date format. I mean, you nearly tried it as a unix timestamp to begin with

matthiasbeyer, 2016-04-10:
Yes, but it is neither Unix timestamp nor any other ... known format. I'd expect at least unix timestamp when seeing such values... but instead I had to ask here how to parse my data...

ogr3, 2016-04-10:
It's a known format. It's a Java standard timestamp (https://docs.oracle.com/javase/7/docs/api/java/lang/System.html#currentTimeMillis())

closed 2016-04-10 by matthiasbeyer