j4velin/Pedometer #83: Someone is selling your code as their own with copyrights removed

vijayku2, 2016-06-01:
http://codecanyon.net/item/pedometer-track-your-steps/16097038

j4velin, 2016-06-01:
Thanks, I contacted them. I guess modifying and redistributing the code is okay with the Apache license, but they should at least mention the source

Aeyoun, 2016-06-17:
“This item is no longer available”

TPS, 2017-02-11:
@vijayku2 @j4velin I guess close this FTW, based on  report from @da2x?

AntonioRedondo, 2017-08-07:
It looks like there is someone else doing it again: https://play.google.com/store/apps/details?id=com.ingic.pacer

This time the culprit behind is an American consultancy company based in San Diego: [INGIC](https://www.ingic.com).

And there is no the first app the steal. [This](https://play.google.com/store/apps/details?id=org.anothermonitor) (mine) also is stolen and uploaded [here](https://play.google.com/store/apps/details?id=com.ingic.deviceanalyzer).

Poussinou, 2017-08-07:
@j4velin Please have a look [here](https://github.com/AntonioRedondo/AnotherMonitor/issues/6) to see all the apps they stolen (there is maybe more but I cannot find the free/open source corresponding app in the F-Droid store).

This [other](https://github.com/SecUSo/privacy-friendly-pedometer/issues/60) open source Pedometer has also been stolen a lot...

Maybe you should take some time (one hour or something so) to check the other apps "Pedometer" on Google Play Store, and report them to Google. I think some of them are just your "Pedometer" with ads in it, and without Apache license, source code and your permission... If you do it, please copy-paste the links of the apps you reported in this thread, so that I can track the "Google developers" who are stealing your work. Thanks

j4velin, 2017-08-12:
Thanks, I wrote them a mail asking to respect the license. If they don't change their store listing, I'll contact the Google Play support next

closed 2017-12-21 by j4velin