j4velin/Pedometer #84: Widget not updating on S7edge

jroldroyd, 2016-06-08:
The subject says it. Installed version 1.4.3 from F-Droid, added a widget, widget remains static at the count it was when added, but touching the widget to open the app shows that the counter has indeed incremented.

j4velin, 2016-06-08:
The widget does not update in real time with every step as this would use
too many system resources

jroldroyd, 2016-06-08:
I must update this. The widget has updated in fact, but it is not syncing frequently. At the moment, the widget value is just under 250 paces behind the value shown when I open the app.

I am not sure what triggers a widget update, but it does not appear to be frequently enough.

j4velin, 2017-04-23:
I'll make the widget update more often soon (see https://github.com/j4velin/Pedometer/issues/97)

closed 2017-04-23 by j4velin