j4velin/Pedometer #85: Number of steps for the current is very high after rebooting

regagain, 2016-06-09:
Hi!

Every time my phone reboots (for example because the battery is too low), the number of steps for the current day is very high. See what happened on Tuesday:
[image]

Is anybody else getting this issue? I am using Pedometer on a Galaxy S6 / Lollipop.

j4velin, 2016-06-09:
Did you "reboot" by correctly shutting down with the power button?

regagain, 2016-06-09:
Ah, no, it happened because the battery was empty.

j4velin, 2016-06-09:
Yeah well, if the device shuts down without notifying the app about the shutdown, then it has no way of persisting the correct data

Aeyoun, 2016-06-17:
I just had this happen too.

I restarted the app using the power button and a normal power off cycle. Had 3500 steps before I restarted, and 22 300 steps after the restart. I then opened the menu and tapped Settings. I didn’t change anything , but click the back button. I now had 49 000 steps. When I walk, the counter increments from 49 000 and upwards,

I use the widget.

Installed the app only three days ago.

dbf256, 2017-01-22:
I have the same shutdown issue, maybe it is possible to store data periodically? I would prefer to loose last 1k steps rather than having +30000 that makes stat for a day useless.

j4velin, 2017-01-23:
The app should save the current step counter every hour already

dbf256, 2017-01-23:
cool, I have this issue yesterday, so I've updated to the latest version and will have a look how it is going with hard resets.

dbf256, 2017-04-02:
Had an issue again with version 1.4.9 today: after battery is empty & device restarted I have 150K steps today.

closed 2017-04-23 by j4velin

dbf256, 2017-04-23:
@j4velin what is closure resolution for this bug - was it fixed in new version or it will not be fixed?

j4velin, 2017-04-23:
I couldn't reproduce this issue on any of my devices so I assume it was fixed at some previous version

dbf256, 2017-04-23:
I have it from time to time at my Nexus 5/Android 6 and I have the latest version. If there could be any useful records in application logs just let me know where they are located and I can send them to you.

j4velin, 2017-05-01:
Feel free to send or upload the logfile if you have the debug version of the app installed (j4velin.de/debug)

nhamer, 2017-08-17:
@j4velin I have it on my Nexus 5 (normal power off sequence, steps go very high), installed the debug build, will upload logs when I get a chance to repro walking around

nhamer, 2017-09-12:
Have been unable to repro since installing the debug build