j4velin/Pedometer #87: Restart counter at midnight

Odaym, 2016-08-10:
Great work on the library, I've tried 3 other apps that attempt this and they are not nearly as accurate when counting, this is really good and almost shake-proof too!

I am not sure where I'm supposed to run the code that will reset the counter to zero when I choose, it gets reset to zero on reboot but I want to reset to zero at midnight and save to shared preferences whatever value I last ended up with, any idea? Thanks again

j4velin, 2016-08-10:
Isn't that exactly what the app already does?

Odaym, 2016-08-10:
why would it do that? there's nothing in the code that mentions that. The only recurrent behavior that Step Counter Sensor is known for is resetting the values at phone reboot, which is why you have the Shutdown and Boot and Power receivers, to do things when that event happens, right?

[http://stackoverflow.com/questions/22649324/step-counter-doesnt-reset-the-step-count](url)

j4velin, 2016-08-10:
Sorry, I'm not sure I understand your question. The app shows how many
steps you have taken each day.  A new day starts at midnight so the app
shows "0 steps" after midnight until you do your next steps. The relevant
code for this is in the database class

Odaym, 2016-08-10:
right, my bad! :D

closed 2016-08-10 by Odaym