j4velin/Pedometer #88: No R.java file in project

dute2, 2016-09-03:
theres no R.java file in your project

Odaym, 2016-09-03:
If you like, watch this video I made about implementing a proper Pedometer on Android https://youtu.be/8Wku0jvnXEA and it goes over this and other libraries that attempt to do this, hope it helps you!

dute2, 2016-09-03:
it is not a solution. i need R.java file for it

Odaym, 2016-09-03:
Good luck

j4velin, 2016-09-04:
R is an automatically generated class when you build the project

closed 2016-09-04 by j4velin

dute2, 2016-09-04:
yeah i know but some files missing . so i need original R.java file of this project

j4velin, 2016-09-04:
Sorry, but that doesnt make sense. If you cant build the app because some files are missing (which?) then adding my R.java wont help

dute2, 2016-09-04:
ok. thanx . actually i am newbie in this field. so more chances are m doing something wrong. but i will be very thankful to you if you just send R.java for it