j4velin/Pedometer #89: Data will be deleted on time zone change

sedrubal, 2016-09-06:
When you travel around the world and your time zone changes, pedometer will delete all data and start from scratch.

liu402883451, 2017-06-26:
how can I solve it ？

Dannysw, 2017-07-28:
I have it acting slightly differently. It just seems to reset the current date, then the next day goes huge. I can have a look at it once I stop travelling.

scrunchy, 2017-08-15:
I've just had this happen when I've traveled between timezones last week - each timezone change threw out the pedometer count for a day or so each side of the time the timezone change occurred.  It doesn't correct when you change back to the original timezone in my experience.

miiichael, 2018-01-26:
I am also experienced this bug.

[image]

I was airborne while crossing midnight, leaving UTC+08 and arriving in UTC+11.

miiichael, 2018-01-26:
Huh, even weirder, my Wednesday stats have now disappeared altogether.
[image]

puddly, 2020-02-25:
I think this problem might be caused by storing only daily counts, but with "today" depending on the local time zone. When you change timezones, "today" changes and weird things happen.

One possible solution is to instead store step counts every 15 minutes, to account for every possible time zone offset. That way, you can always compute the "daily" step count based on the user's current timezone without making the database dependent on it. The DB size will grow much faster, but you'd still need to walk nonstop for 30 years before it hits a million rows.

Your displayed step counts would change if you travel between timezones, but that's something that can be worked around if you also store the current timezone in the database for each row.

@j4velin what do you think?

j4velin, 2026-09-26:
Fixed in c145aff: a time zone change keeps the day's existing entry (the one within 12 hours of the new local midnight) instead of starting a new one, and no data is removed.

This is on master and will ship with version 2.0.0.

closed 2026-09-26 by j4velin