j4velin/Pedometer #91: Lollopop api

atif-afridi, 2016-11-09:
hey there first of all nice work and secondly i want to ask how to run on lollipop its not working there giving error .. this app requires a dedicated hardware step sensor

Odaym, 2016-11-09:
what error? It works on Lollipop just fine, the hardware sensor required for this to work is at a min of 4.4 (Kitkat)

atif-afridi, 2016-11-09:
Check it here is the screenshot

Odaym, 2016-11-09:
in this video I work on this pedometer and compare with others, there is only 1 paragraph needed to get this whole thing running https://www.youtube.com/watch?v=8Wku0jvnXEA

atif-afridi, 2016-11-09:
its not the project i am working on plz check this source code i have issue
in this
https://github.com/j4velin/Pedometer

sure i ll watch the whole video but now i have to make this working
best regards

j4velin, 2016-11-09:
Well, if the app shows the error saying it required a dedicated step sensor, then that means that you device does not have a dedicated step sensor but the app requires such one...

closed 2016-11-09 by j4velin

atif-afridi, 2016-11-10:
ok but how to solve this issue i mean if i don't i have the hardware there must b any alternative or not ?
thanks