j4velin/Pedometer #99: Old version on F-Droid

sniperle, 2017-04-23:
The version on F-Droid is outdated.
As of today (2017-04-23) it's still 1.4.3

j4velin, 2017-04-23:
Sorry, but you need to contact the F-Droid guys for that. I don't build or upload anything to F-Droid

closed 2017-04-23 by j4velin

sniperle, 2017-04-23:
Ah OK, then I will have a look at the F-Droid page. 

Thought you'd be interested in that the latest version is available on F-Droid, since you use the "Get in on F-Droid" Download button on your site.

mimi89999, 2018-03-30:
@j4velin Please tag all feature releases so that the F-Droid bot could automatically update the metadata file.

MarcRdC, 2018-11-06:
@j4velin Last advice from @mimi89999 still holds. If you tag your releases here on GitHub, they will be automatically detected and built by F-Droid. Thank you!

j4velin, 2018-11-06:
I'll try to keep that in mind