j4velin/Pedometer #116: Incorrect entry for a day when the phone does a soft reboot

Akendo, 2018-02-20:
Hello,

Thank you very much for this software. I found a bug during my daily usage. When I do a soft reboot of the phone the step count raised incorrectly by a big amount. From almost zero to almost 10.000 steps.

I used App Version 1.5.1 on a Samsung Galaxy S7 Edge with Android 7.1.2 (Lineage: 14.1)

sgaillard, 2018-04-17:
Hello,
First thanks for that app witch is one of the best pedometer i've tryed, one of the less overcounting!
Faced the exact same behavior with version 1.5.4 from Google Store on Android 7.0 (Huawei P9 lite VNS-L31). It seems to be related to the display of importated data.
Soft reboot set the current date to around 10.000 steps as well as importing data from an export.
Tryed with initial count on 0 or with already counted steps for the current date.
Thing strange, if you export the data, counts are OK in the exported file.
So might be only a display bug.
Best
Sylvain

j4velin, 2018-04-17:
What do you mean with "soft reboot" exactly?

um 22:41 Uhr:

sgaillard, 2018-04-17:
I mean reboot with the "normal" process : press the button for 3 seconds on the phone then select reboot and validate. Not a hard shuttdown pressing the button for 10 seconds and then switch on the phone again witch obviously leads to data loss for what I've understand but not tested.

kijin, 2018-04-23:
I have the same issue in a Samsung Galaxy A3 2016, but the counter changes back to the correct value after taking a few steps.