j4velin/Pedometer #131: Android System thinks pedometer is running in the foreground

lwmv, 2018-11-08:
I'm running LineageOS 15.1, Android 8.1. I updated Pedometer from v1.5.6 to v1.5.8 today. The change log of v1.5.8 reads "add settings shortcut to notification settings on Android 8+". Now in the settings of pedometer there is a "Notification settings" option, I click on this option and it will jump to the system notification settings of pedometer. 

Now the problem is, if I turn off the notification of Pedometer, the Android System will show a notification saying that "Pedometer is using battery". And in the battery usage information of Pedometer, the "While in background" entry is 0 minute, the “While in active use" entry equals the time I unplug my phone after full charge. That is to say, though pedometer is actually running in the background, the Android System thinks Pedometer is running in the foreground (in active use) and pop up a notification notifying me of that.

I downgrade to v1.5.6 and the issue is gone. Now Pedometer is being deemed by the Android System as being running in the background again, and the "While in background" entry starts to grow normally. In the settings of v1.5.6, the "Show notification" is a check box, instead of an option that will direct me to the system notification settings of Android 8.1.

[image]
[image]
[image]

j4velin, 2018-11-08:
With the target SDK now set to 8.0, the app needs to have a "foreground
service" to be kept alive. And foreground service require a notification
for some reason...

lwmv, 2018-11-15:
OK, I will just ignore the "Pedometer is using battery" notification.

artfulrobot, 2019-05-28:
I realise it's not this app's fault, it's done hideous Android decision, but I down graded to an earlier version of this app to avoid the notification confusion and top bar clutter that the latest version brings.